autowarefoundation/odd-usecase-scenario #4: [SCENARIO-FAILED] UC-NTR-001-0001_case(1|2) - Failed with excess deceleration (Non-deterministic case)

kohtake, 2025-04-28:
### Test Information

| Scenario | Suite | Catalog |
|----------|-------|---------|
[UC-NTR-001-0001_case1](https://evaluation.tier4.jp/evaluation/scenarios/0c51ed83-bffa-419e-9725-c649fe4780f3?project_id=awf) | [Dense Urban](https://evaluation.tier4.jp/evaluation/suites/05e1818a-85a8-4940-afb1-9e1795fe7b7f?project_id=awf) |  [Urban-Dense](https://evaluation.tier4.jp/evaluation/vehicle_catalogs/aaf96902-ab17-4957-93a0-e3b6aa7b9efd?project_id=awf)| 
| [UC-NTR-001-0001_case2](https://evaluation.tier4.jp/evaluation/scenarios/70003386-d425-48ed-8746-7a2368d53b67?project_id=awf) | [Dense Urban](https://evaluation.tier4.jp/evaluation/suites/05e1818a-85a8-4940-afb1-9e1795fe7b7f?project_id=awf) | [Urban-Dense](https://evaluation.tier4.jp/evaluation/vehicle_catalogs/aaf96902-ab17-4957-93a0-e3b6aa7b9efd?project_id=awf) |

### Test Runs

| Test Week | Test Run Link | Failure Rate | Source Version | Notes |
|-----------|---------------|--------------|----------------|-------|
| 2025-04-13 | [2275a72b](https://evaluation.tier4.jp/evaluation/reports/3331050b-5ee1-52e3-9fbd-ad2d696b767a/tests/2275a72b-d74d-5b7a-9197-729c1520bd4b?project_id=awf) | 2/6 | [ea21db7c](https://github.com/autowarefoundation/autoware/commit/ea21db7cf358c2ba3cfe735685a20ae2bd253b5d) | Initial failure |

### Failure Description

1 ego moved before NPC finishes overtaking + excessive deceleration (Non-deterministic)

### Affected Repositories

- [ ] [odd-usecase-scenario](https://github.com/autowarefoundation/odd-usecase-scenario)
- [ ] [autoware_universe](https://github.com/autowarefoundation/autoware_universe)
- [ ] [autoware-projects](https://github.com/autowarefoundation/autoware-projects)
- [ ] Other repositories (specify below):

diegoyabuki, 2025-06-09:
First we need to determine whether the correct deceleration limit is employed in the scenario.
Given the scenario itself, it seems that the usage of higher deceleration is required to avoid collision with the NPC.

diegoyabuki, 2025-06-10:
All cases failing with this error happen when ego must brake to prevent a collision.
The collision situation happens due to other problems either in Autoware (feature for waiting for NPC overtake is not available) or the scenario (cyclist NPC drives into the ego vehicle). 
In such cases, it could be argued that the system should use the highest deceleration available to prevent the collision, so this issue may not be relevant.