autowarefoundation/odd-usecase-scenario #32: Prepare a local environment using CARLA 0.10.0 (UE5)

diegoyabuki, 2025-06-04:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

Set up CARLA in a local machine.

### Purpose

A local environment is required to enable trying out the CARLA features. This task involves preparing such environment.

### Possible approaches

- Follow the CARLA documentation to install CARLA in the local machine. Preferably build from source, in case we need to add modifications at a later point.
- In case troubleshooting is needed, our colleagues from the National Taiwan University might be able to help.

### Definition of done

- CARLA 0.10.0 is installed and running on a local machine.
- No error messages are shown during the simulation.

maxime-clem, 2025-06-17:
The document is private. Can it be made public ?

diegoyabuki, 2025-06-17:
Definitely! Since I can't move files out of AWF Google Drives, I created a copy in our public drive. Here is the new [link](https://docs.google.com/document/d/1_idYydgOf7s2W9oNFAaYIeOX6sYDLOMzrCzWI5ZsVDA/edit?usp=drive_link).

maxime-clem, 2025-06-17:
Thank you very much 🙇